sourceofvegemite/fan_controller #3: Exceptionally low D1 reverse current

JulianKeledjian, 2025-01-25:
### Description

D1 reverse currents are not rated sufficiently high to dissipate high current events. 

### Screenshots

[image]

### Files

Top level schematic.

### Tasks

Issue may be remediated by either:

- [ ] Replacing D1 with either a TVS or Zener.

JulianKeledjian, 2025-01-25:
Also, I think the diode part number is incorrect? I can't find any references to PEGM3030EX.

sourceofvegemite, 2025-01-26:
## Action

Diode part number and schematic symbol found to be incorrect.
Component is chosen as a flyback/snubber diode, over-voltage protection.
New part researched, B340-13-F Schottky chosen instead.

Investigating the need for a separate over-voltage protection (OVP) circuit.

## Screenshot

[image]

closed 2025-02-23 by sourceofvegemite