sourceofvegemite/fan_controller #5: Insufficient decoupling for U9 reference

JulianKeledjian, 2025-01-25:
### Description

U9 does not require output capacitance to be stable, however, it is highly recommended for both noise shaping and decoupling reasons. 

### Screenshot

[image]

### Tasks

- [ ] Increase C38 to 10nF or 100nF

sourceofvegemite, 2025-01-26:
## Comment

Unsure whether C38 is the correct capacitor to be addressed.
Screenshot above suggests that an output capacitor is not required for the reference voltage circuit, however recommended output capacitor is given between the 0-1nF range.
It then suggests that an input capacitor of 2.2uF be included.

ADC component suggests more aggressive supply decoupling, shown in screenshots. Open to suggestions as to the discrepancy.

## Actions

1. Input capacitors for all MCP1501-30xCH parts increased from 100nF to 2.2uF as suggested by text.

## Screenshots
**ADC suggested decoupling.**
[image]

sourceofvegemite, 2025-02-22:
## Comment

Buffered input application shows use of an LM4132, which is also a precision voltage reference. The datasheet for the LM4132 shows an optional output capacitor, whereas the ADC081C021 datasheet shows two output capacitors. Decided to follow the ADC081C021 datasheet.

## Action

U9 now has two output capacitors, 4.7uF and 100nF respectively.
U10 now has two output capacitors, 10uF tantalum and 100n ceramic respectively. Chosen due to recommendation of U11 datasheet (DAC).

closed 2025-02-22 by sourceofvegemite