sourceofvegemite/fan_controller #8: Protection diode forward biased upon fan idle and powerup

JulianKeledjian, 2025-01-25:
### Description

The peak detector will instantly charge to 3v3. This will be on the cusp of forward biasing the protection diodes both internally- and externally on the U12 ADC.

### Screenshot

[image]
[image]

JulianKeledjian, 2025-01-25:
@sourceofvegemite Another issue is that the charge-discharge times will be highly asymmetric which may effect any closed loop control. Does this need to be another issue?

sourceofvegemite, 2025-01-26:
## Comment
Acknowledged that C30 will instantly reach +3v3 upon start-up.

Asymmetry in charge-discharge times deliberate due asymmetry in switching of Q7. 'On' time of Q7 is on order of milli-seconds.

## Action
Changed D4 to 1N4148 to achieve larger VF. BAT43W estimated VF is 0.26V worst case.

## Screenshots
**BAT43W VF from datasheet.**
[image]

**1N4148 VF from datasheet.**
[image]

closed 2025-02-22 by sourceofvegemite